PE5PVB/TEF6686_ESP32 #597: RC14: Still mixed up EON PS text fragments

andimik, 2024-06-05:
There is still a bug in EON PS. It will show fragments (mostly 2 bytes) of another PS.

It will disappear after 2nd time the PS was received.

PE5PVB, 2024-06-05:
Yes I know. For some reason it sometimes just doesn't receive all bytes every run. Can't do anything about that.

andimik, 2024-06-05:
This is strange. I cannot remember this was the case before all RCs ...

PE5PVB, 2024-06-05:
I have no idea how to fix it. Things have to do with sorting and showing the EON data with some kind of refresh delay, otherwise the ticker won't work. I don't think it's a real problem, because it will auto correct.

andimik, 2024-06-05:
But I found a TEF with version 2.00.15 and this does NOT show wrong PS texts at all.

Please note that I'm not talking about missing characters, but about mixed up characters.

PE5PVB, 2024-06-05:
I know but that caused other issues. The problem will only occur just when tuning in and EON list is still building and will auto solve. 

I found one thing which could cause this kind of issues and applied it. Maybe it will help, otherwise I have no solution.

andimik, 2024-06-08:
Seems this has been solved in repository, have not seen that after 38bdf2ab07ad3cefbbf6db3dd17b765155db0747

closed 2024-06-08 by andimik