PE5PVB/TEF6686_ESP32 #601: OnScreen Battery Indicator not showing any value

KmanOz, 2024-06-08:
Battery has been turned on from main menu to show voltage or percentage and shows nothing. Not sure if this is something to do with the Chinese variant I have purchased. I have set the model selector to base.

For reference this is the model I have purchased

https://www.aliexpress.com/item/1005006676689539.html?spm=a2g0o.order_list.order_list_main.16.7d011802H1fE7F

andimik, 2024-06-08:
Could you add a picture?

KmanOz, 2024-06-08:
[image]
[image]

PE5PVB, 2024-06-08:
You can't show battery level while connecting to WiFi. This is a hardware limitation.

andimik, 2024-06-08:
@KmanOz 
That's why I asked you to post a picture, so please close it. This is normal behavior.

KmanOz, 2024-06-08:
Fancy

KmanOz, 2024-06-08:
No worries cheers

closed 2024-06-08 by KmanOz

PE5PVB, 2024-06-08:
Blame the Chinese for choosing a wrong gpio

KmanOz, 2024-06-08:
Fair enough. I blame them for everything. Nothing against you brother. Keep up the good work.