PE5PVB/TEF6686_ESP32 #605: After retuning, the AF still "searches" for another frequency when the screen is off

Mike86-CZE, 2024-06-08:
If I turn off the screen and AF retune to a better frequency, the radio still "searches" every second (corrupts the sound).  when i turn on the screen the problem stops.

PE5PVB, 2024-06-08:
I think I've fixed this.

Mike86-CZE, 2024-06-08:
this is what happens with RC14.

PE5PVB, 2024-06-08:
See 6c36b08

Mike86-CZE, 2024-06-08:
Ok, I will test in the next version.

numerci, 2024-06-10:
On RC15 for me it works exactly the same as on RC14. Same sound problems when switching the station with the screen off. For me at least there is no change.

PE5PVB, 2024-06-10:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/58ee9fabee159f7f21d5fb07d670c7c2c3e9cd6a

numerci, 2024-06-12:
On RC16 it finally works fine. At least for me.

Mike86-CZE, 2024-06-13:
Now it works properly.  I'm closing this topic.

closed 2024-06-13 by Mike86-CZE