PE5PVB/TEF6686_ESP32 #607: No sound during SW auto tuning after turning on the screen

Mike86-CZE, 2024-06-08:
If I turn off the display in the SW band and start automatic tuning with the encoder and turn on the display during tuning, the tuning stops at the current frequency, but the sound remains muted.

Mike86-CZE, 2024-06-08:
And I remind you that the beep still does not work for SW bands when the screen is off.  It was already fixed once, but the problem reappeared over time.

PE5PVB, 2024-06-08:
I can't reproduce this problem. Please let me know which menu settings I should have and then a step-by-step plan to reproduce.

Mike86-CZE, 2024-06-08:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

PE5PVB, 2024-06-08:
OK. And how to reproduce?

Mike86-CZE, 2024-06-08:
1. switch to the SW band
2. set the tuning mode to AUTO
3. turn off the display by holding down the BAND button 4. turn the encoder to start tuning 
5. before a station has time to tune in, turn on the display with the BAND button 
6. the tuning has stopped at some frequency, but the sound  did not activate

Mike86-CZE, 2024-06-08:
The correct behavior would be to not stop tuning at all after turning on the display and continue to search for a station.

PE5PVB, 2024-06-08:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/646460306b1b57c3f05238d339334d6ad5ce674f

numerci, 2024-06-11:
This seem to be working fine for me now on RC15

closed 2024-06-12 by PE5PVB