PE5PVB/TEF6686_ESP32 #609: Repeating a short radiotext

Mike86-CZE, 2024-06-09:
If the station does not broadcast a series of spaces after the text in the radiotext and the text is shorter than a line of the screen, the text is unnecessarily repeated and scrolled.

https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/5f4452d2-7600-4db4-892c-59d2c4fba66c

Mike86-CZE, 2024-06-09:
But now I'm seriously not sure if they started broadcasting the text duplicated... Maybe it would be good to have optional radio text start and end markers?

Mike86-CZE, 2024-06-09:
No, they really only broadcast the text once.
[image]

numerci, 2024-06-09:
Would be possible to also put a screen shot with the RDS settings from the Menu? For me this works fine (for stations transmitting the end of line and also for the ones that do not). There has to be a difference in the settings that could be important.

Mike86-CZE, 2024-06-09:
[image]

Mike86-CZE, 2024-06-09:
It is strange that the radiotext is duplicated on the SONY XDR-S41.  So I'm not really sure how it's broadcast.

numerci, 2024-06-09:
Thanks.... we have the following differences in the setup:
RDS Filter = ON
Fast PS = OFF
Show clock = OFF
I will give it a try with your setup.
PS: Please note that RDS Filter now works exactly the opposite as before (as designed).

numerci, 2024-06-09:
Tested with your settings on the stations I have with end of line and without and it works fine. The text is not duplicated for me. It doesn't mean there is no defect just that these settings have no influence.

Mike86-CZE, 2024-06-09:
On a Samsung Galaxy Mini running Android 2.3.6 with the Spirit app, the radiotext is also displayed twice.  So maybe it's just newer Android that artificially removed the duplication, which confused me.

Mike86-CZE, 2024-06-09:
So it looks like it's not a bug.  But it would probably be useful to have a tag at the end of the radiotext.  I am closing this issue.

closed 2024-06-09 by Mike86-CZE

PE5PVB, 2024-06-09:
Still curious about the RDS data of this station. Can you make a log with RDSSpy or a MPX recording?

Mike86-CZE, 2024-06-09:
A 192kHz sound card is required for MPX recording, right?

andimik, 2024-06-09:
Or with SDR#

PE5PVB, 2024-06-09:
Output of RDSSpy is also ok.

Mike86-CZE, 2024-06-09:
[fajn927.zip](https://github.com/user-attachments/files/15751600/fajn927.zip)

Mike86-CZE, 2024-06-09:
> So it looks like it's not a bug. But it would probably be useful to have a tag at the end of the radiotext. I am closing this issue.

A tag is probably not needed.  Just turn on the underscores.

andimik, 2024-06-09:
The text is already repeated on encoder side

```
andreas@E535:/tmp$ cat ./fajn927.spy | redsea -i hex | grep text
{"pi":"0x2335","group":"2A","radiotext":"FAJN RADIO - PROSTE HITY        FAJN RADIO - PROSTE HITY","prog_type":"Pop music","tp":true}
```

andimik, 2024-06-09:
[image]

PE5PVB, 2024-06-09:
Yup. Nothing to fix here. It's just a crazy engineer.