PE5PVB/TEF6686_ESP32 #611: 0.1Mhz step when scanning

numerci, 2024-06-09:
This is more a question than possible a bug. What is the expected behavior when running an Automatic scan with a limit set for FM band from 65Mhz to 108Mhz. Should the firmware know to use 0.03Mhz or the range 65-74Mhz? Currently it uses 0.1 on the entire range. This affects also the AutoScanner plug in if the user manually sets a frequency between 64 and 74 Mhz and starts an automatic scan.

Note: If the user selects OIRT on the TEF it self it works well but what if the user extends the FM band to also cover OIRT.

PE5PVB, 2024-06-09:
I already fixed this. When you first switch to OIRT frequency and start the scanner is should use the right stepsize. Problem with the webserver is the lack for communication between the radio and the webserver on this part.

numerci, 2024-06-10:
On RC15 it works as before. At least for the Autoscan on TEF.

numerci, 2024-06-10:
Update: actually now from the webserver is working but from the TEF itself is still not working as stated in the previous comment. Question is if we really need it to work from the TEF....

PE5PVB, 2024-06-10:
It will work when you switch to OIRT. But if you set the bandlimits of FM to OIRT range and stay on FM it won't change the stepsize. This is no bug.

numerci, 2024-06-10:
Ok, I do not have anything against this resolution. Bug closed. :)

closed 2024-06-10 by numerci