PE5PVB/TEF6686_ESP32 #622: Feature request: Increase the Set Volume interval from 10dB to 20 or 30

numerci, 2024-07-02:
As the latest chinese TEF's have strange audio levels would be very helpful to have this interval increased. 10dB doesn't see to be enough anymore. :( Hope 20 or 30 would be enough. If there is no hardware limitation for this.

PE5PVB, 2024-07-03:
Range of the chip is limited to -60 - +24dB. Please note that a setting different to 0 will have effect on audio quality.

numerci, 2024-07-03:
You may be right about the audio quality when the audio is normal on a device but when is already working strangely it may be an improvement. Can it be added just as an experiment for one of the next versions? If it has unwanted effects we can go back to the previous range.

closed 2024-07-27 by numerci