PE5PVB/TEF6686_ESP32 #625: Prevent PI scanning from triggering when RDS reception is lost on stations without AF.

Yurij20, 2024-07-05:
I analyzed the operation of the function more deeply and noticed that PI scanning is triggered not only when switching to a preset with no reception, but also when RDS reception is lost on stations without AF.  I consider this case to be unnecessary for scanning to trigger and should be excluded.

PE5PVB, 2024-08-03:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/eba22aa41ad2045e874856a8e9eac52fbb3b4b9b

Yurij20, 2024-08-04:
> Fixed in [eba22aa](https://github.com/PE5PVB/TEF6686_ESP32/commit/eba22aa41ad2045e874856a8e9eac52fbb3b4b9b)

I checked that the pi search triggering after exiting the MEM mode has been eliminated, but the pi search triggering still remains when RDS reception is lost on stations without AF when using the MEM mode.      Here I think it is necessary to modify the moment so that the search by PI works at the moment when you switch to a saved station and at that moment there is no reception on it.  But if the RDS carrier has already been received, then there should be no more PI search triggers until you switch from this station.

PE5PVB, 2024-08-05:
Check https://github.com/PE5PVB/TEF6686_ESP32/commit/edcccf78029b7c6a933259b166f6dea858cd8dc1

closed 2024-08-05 by PE5PVB