PE5PVB/TEF6686_ESP32 #636: FMDX scanning issue

lvxc88, 2024-08-18:
1. When the frequency band of the starting channel is not FM and there is no FM channel within the selection range, FMDX scanning may sometimes get stuck
2. When FMDX scans, some online commands may be abnormal, such as the T command to switch a frequency, which will not be able to stay on that frequency

lvxc88, 2024-08-22:
M命令和T命令结果在频率扫描的时候可能都有点异常，我觉得可以在这两个命令执行之前停止频率扫描

lvxc88, 2024-08-22:
The results of the M and T commands may be slightly abnormal during frequency scanning. I think we can stop frequency scanning before executing these two commands

PE5PVB, 2024-08-31:
Hi, point 2 is fixed in 10fefe2

Can you give me a step-by-step guide to reproduce point 1?

lvxc88, 2024-08-31:
gui.cpp line 4549: if (presets[scanstart].band == BAND_FM && presets[scanstart].frequency == EE_PRESETS_FREQUENCY) {

1. Select channel scanning only
2. When the starting channel stores an AM frequency or the starting channel frequency is 0, this if statement will not work and the scan will continue.
3. Then when all channels within the scanning range are AM, the scanning will continue to stop at the first channel, giving the feeling of being stuck

lvxc88, 2024-08-31:
Also, may I ask if it is necessary to add "if (scandxmode) cancelDXScan();" in line 144 of the comms. cpp file?

PE5PVB, 2024-08-31:
> Also, may I ask if it is necessary to add "if (scandxmode) cancelDXScan();" in line 144 of the comms. cpp file?

Thanks. I've forgotten that one. https://github.com/PE5PVB/TEF6686_ESP32/commits/main/

PE5PVB, 2024-08-31:
I have to think about a way to make a error handler for this.

lvxc88, 2024-09-01:
I think it is possible to conduct a comprehensive inspection of the channels within the scanning range before scanning. At least two FM radio stations need to be scanned, and the inspection speed is not slow. In addition, OIRT needs to be uniformly processed. If it is supported, then during the inspection, it is not enough to only check FM radio stations

closed 2024-09-21 by lvxc88