PE5PVB/TEF6686_ESP32 #639: DX Scan and automute

numerci, 2024-08-31:
Preconditions on 2.11.4: 
1. Mute audio while scanning - ON
2. Auto cancel scan - OFF
3. Memory channels - Exclude
Steps: 
1. Start DX Scan
Result: The entire time during scan the audio is muted. Even when it finds signal on a station and it stops.
Expected: When a signal is found and the scan pauses the audio should be switched on.

PE5PVB, 2024-08-31:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/1199e3cf5980dec4f6e405edfa420506c4969baa

lvxc88, 2024-09-01:
I think it is possible to conduct a comprehensive inspection of the channels within the scanning range before scanning. At least 2 FM radio stations need to be scanned, and the inspection speed is not slow. It is for reference only

numerci, 2024-09-01:
@lvxc88 wouldn't be much easier if you would share the mod you did with the community? Sjef is re-implementing things you already done. He could focus on doing new stuff.

lvxc88, 2024-09-01:
> @lvxc88 wouldn't be much easier if you would share the mod you did with the community? Sjef is re-implementing things you already done. He could focus on doing new stuff.

I actually considered merging it from the beginning, but due to my unfamiliarity with GitHub, I had already refactored a lot before completing it. After completing it, I was unable to merge it. Therefore, I am considering opening it up on GitHub in the future. However, there are still some other reasons that cannot be carried out temporarily. Please understand! In addition, PE5PVB is designed to create more in-depth features in a superior way, rather than duplicating work. I have always regarded him as my teacher.

lvxc88, 2024-09-01:
> I think it is possible to conduct a comprehensive inspection of the channels within the scanning range before scanning. At least 2 FM radio stations need to be scanned, and the inspection speed is not slow. It is for reference only

> @lvxc88 wouldn't be much easier if you would share the mod you did with the community? Sjef is re-implementing things you already done. He could focus on doing new stuff.

> @lvxc88 wouldn't be much easier if you would share the mod you did with the community? Sjef is re-implementing things you already done. He could focus on doing new stuff.

I'm very sorry!I have written the reply to another question here

numerci, 2024-09-02:
Partially fixed on 2.11.5. Will add a bug with the remaining case. This one is closed.

closed 2024-09-02 by numerci