PE5PVB/TEF6686_ESP32 #653: Please move the beginning of the red marker of the signal and modulation meter bands

Yurij20, 2024-09-01:
Please move the beginning of the red marker of the signal and modulation meter bands to the positions indicated by me in the figure.  I think it would be more correct to display the inflated level)
[image]

PE5PVB, 2024-09-17:
What is the reason for asking this?

SpiderBlocker, 2024-11-11:
I believe it is commonly accepted (and even standardized) that S9 on a receiver marks the upper limit, above which stronger signals are to be indicated  with an additional +dB-rating, so that would match the current implementation. As far as modulation index is concerned, I am not sure, but aesthetically I would prefer it to line up evenly with the S-meter.

PE5PVB, 2025-01-06:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/591680fedeb625981056252c38ef15ba6ebcdf5c

Yurij20, 2025-01-11:
> > Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/591680fedeb625981056252c38ef15ba6ebcdf5c

Thank you for accepting my offer)

closed 2025-01-11 by Yurij20