PE5PVB/TEF6686_ESP32 #660: Code compilation problem.

mi-cha-el-80, 2024-09-07:
Hello to all forum users.

I am very interested in the project of this radio with the TEF6686 system. I have purchased all the necessary elements for its first implementation, but I have a problem with compiling the code. I have used the available instructions but unfortunately I have not found a solution to my problem.
During compilation an error appears:

C:\Users\Majkel\Documents\Arduino\libraries\libraries\TEF6686_ESP32-main (7)\src\TEF6686_ESP32\TEF6686_ESP32.ino:11:10: fatal error: src/WiFiConnect.h: No such file or directory
11 | #include "src/WiFiConnect.h"
| ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: src/WiFiConnect.h: No such file or directory

Can someone help me solve this problem?
Regards.

[image]

PE5PVB, 2024-09-07:
Looks like something goes wrong with the files and subdirectories. Make sure they match Github. So you have the INO file in a subdirectory, and that subdirectory has another subdirectory called src with all the files inside.

lvxc88, 2024-09-07:
Check if the project folder address is set in the preferences

mi-cha-el-80, 2024-09-08:
Hello everyone again.

Thank you for your interest in my topic and help. The problem was actually in the file structure where the sketches are stored, by trial and error I managed to solve the problem and go a step further, unfortunately this is not the end of the problems, this time such an error:

Compilation error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'

Can I ask for help in solving this problem again?

Thank you.

[image]

PE5PVB, 2024-09-08:
Please use latest ESP32 boardmanager v2.x.x

V3 isn't compatible yet

closed 2024-09-17 by PE5PVB