PE5PVB/TEF6686_ESP32 #665: Modulation Meter

rf-oldie, 2024-09-29:
What's the unit of the FM modulation bar? I guess "kHz", because the red range starts at 70.
A 50 kHz modulated signal is shown with a value of about 70. This matches with the datasheet's 100 = 75 kHz. 
So for correct modulation indicator the MStatus should be multipied by .75

PE5PVB, 2024-10-01:
Well, it can't be compared to deviation, it's just audio level. With deviation you should be able to measure levels above 15kHz, which the meter doesn't. So in the end it's just an indication.

rf-oldie, 2024-10-01:
Ok, if it's derived from audio, it's clear.
Thought you were using the "Get_Quality_Status" modulation percentage for that, didn't check that.

closed 2024-11-07 by PE5PVB