PE5PVB/TEF6686_ESP32 #666: 2.11.6: Mute still on if signal detected on DX Scan

numerci, 2024-10-13:
Preconditions:
Wait on signal = off
Mute audio while scanning = on
Auto cancel scan = off
1. Start DX scan (either from Menu or from the button).
2. Wait till a station is found.
Restul: The RDS loads but the adio is still off.
Expected result: When signal detected the Mute should be switched off.

Note: if Auto cancel scan is on then it works fine.

PE5PVB, 2024-10-13:
Yes, that is correct. Because this is more for unattended use.

numerci, 2024-10-13:
I would assume this is new because on 2.11 it used to word differently.

closed 2024-10-13 by numerci

PE5PVB, 2024-10-13:
Test it with wait time on signal only ON/OFF

numerci, 2024-10-14:
This I have already done and it was working fine. Only with equal stop time was not working anymore like on 2.11. I thought you want to keep the old functionality too.