PE5PVB/TEF6686_ESP32 #668: Advanced RDS EON screen bugs

mrwish7, 2024-10-16:
Not sure how long this one has been the case, but I could replicate it in both of the latest two portable releases (2.11.5 and 2.11.6).

In the advanced RDS screen with EON details (screen 2/n), there are some bugs -

- Each PTY gets shown as 254, and then if a correct PTY is received, 254 does not get cleared, the correct PTY just writes over the top of it on the display.
- Where there are a lot of EON networks, the list overflows the container off the bottom of the screen, even though a page 3 also gets triggered (as expected).
- In this case - when trying to access page 3, the radio crashes and reboots.

https://github.com/user-attachments/assets/027ffa70-a9e1-49f9-95ea-fd12b36037d7

PE5PVB, 2024-10-24:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/419db542d94e0c4614df647a04b11510c7b103e3

PE5PVB, 2024-11-07:
Please confirm

lxsxl, 2024-11-07:
The issues mentiones above are fixed. But there is still an issue if there are three EON frequencies (see photo). Btw: Why is the label "MF" shouldn't it be "AF" or "EON-AF"?
[image]

andimik, 2024-11-08:
Seems like this stands for multiple  frequencies, right?

PE5PVB, 2024-11-08:
> The issues mentiones above are fixed. But there is still an issue if there are three EON frequencies (see photo). Btw: Why is the label "MF" shouldn't it be "AF" or "EON-AF"? ![IMG_1071](https://private-user-images.githubusercontent.com/140750201/384136734-bb95451a-84c2-46cb-93de-420ffd8b8f90.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEwNzYzMDAsIm5iZiI6MTczMTA3NjAwMCwicGF0aCI6Ii8xNDA3NTAyMDEvMzg0MTM2NzM0LWJiOTU0NTFhLTg0YzItNDZjYi05M2RlLTQyMGZmZDhiOGY5MC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwOFQxNDI2NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00YjZmNTIzZThjNzEwNGMxNTU3NjVhYTNlOTFjMGViYzM1MGRiMzY3OTM0NTIzYTdiN2EzZTk3MTJjNWM4ZGNjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.SlAxhmaYhDcQRg4r0TPaSOxziH8o9b2CxLFNdKCrTEA)

At which frequency is this?

lxsxl, 2024-11-08:
What do you mean exactly? TEF is tuned to 91,8 MHz (as you can see in the foot row). I think I sent you a recording in the past, so you should be able to reconstruct it. If you need another recording, please let me know.

lxsxl, 2024-11-08:
> Seems like this stands for multiple frequencies, right?

Idk, the label MF is also used if there is only one single EON-AF...

PE5PVB, 2024-12-09:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/a4df21da65d7f3b9029258c8afb09932d2fedc21

lxsxl, 2024-12-10:
There are still errors in the EON view: Seems that it is only when the sorting has to be rearranged, then the old data (at least MF and PTY) are kept in the rows where they were displayed first as long as the right data are not received.

[image]

PE5PVB, 2024-12-11:
> There are still errors in the EON view: Seems that it is only when the sorting has to be rearranged, then the old data (at least MF and PTY) are kept in the rows where they were displayed first as long as the right data are not received.
> 
> ![IMG_1272](https://private-user-images.githubusercontent.com/140750201/394457425-34b96720-7543-48bc-b15b-7df09d277e14.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5MDgyNjgsIm5iZiI6MTczMzkwNzk2OCwicGF0aCI6Ii8xNDA3NTAyMDEvMzk0NDU3NDI1LTM0Yjk2NzIwLTc1NDMtNDhiYy1iMTViLTdkZjA5ZDI3N2UxNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxMVQwOTA2MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03Y2UwNjk0MDQzNmJlNjNiMWJjZjE5MTFhY2RkMjc4NmE5YzZmODY4MzVlYTY3YzAwN2E4YWVlZTgyODFkZGM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.wayBMIZyroHjXXTmSE0X3ai_MZ2qhBy_TutbHnCzVuk)

Is it OK when you leave and re-open the EON view?

mrwish7, 2024-12-11:
Cycling back through the screens (e.g back to page 1 then onto page 2 again) clears the overlaps and makes it display clearly again.

The duplicated frequencies and PTY overlaps only show if you wait on screen 2 while the data loads and sorts.

PE5PVB, 2024-12-11:
Thanks. I'll look into it.

PE5PVB, 2024-12-18:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/d4f1fec09d28b532746dbff0c0f83a7cde6039ea

lxsxl, 2024-12-19:
Looks fine again, thanks for fixing!
In my opinion this issue may be closed.

closed 2024-12-19 by PE5PVB