PE5PVB/TEF6686_ESP32 #669: Clock displaying in UTC +01:00

AaronJowitt, 2024-10-20:
v 2.11.6

PE5PVB, 2024-10-21:
Please explain

AaronJowitt, 2024-10-21:
I'm in UTC +10:00. The clock used to display in local time zone with earlier fw versions.

It's currently 06:55 local. 
On a TEF with v2.11.1, clock displays 06:55 (UTC +10:00) . 
On a TEF with v2.11.6, clock displays 22:55 (UTC +01:00).

I've tuned to at least half a dozen stations that broadcast CT (including one that has CT wrong by about 10 minutes), and on all the clock time displayed is in UTC +01:00

andimik, 2024-10-22:
@kb8u-gqrx

Could you please check your latest commit?

AaronJowitt, 2024-10-25:
v2.11.7 does not show a clock.  
Clock is ON in display settings.

kb8u, 2024-10-26:
I found a related bug and have submitted a fix.  Sorry I missed this, I have no practical way of testing this since I'm in UTC-4 time zone.

AaronJowitt, 2024-11-07:
Thanks for the fix.  The clock is back in v 2.11.8

closed 2024-11-07 by PE5PVB