PE5PVB/TEF6686_ESP32 #674: xdr-gtk v1.2 problem w/Spectral scan display

kb7wox, 2024-10-29:
With a Deepelec DP666, or a silver TEF model the XDR-GTK v1.2 will not display the Spectral scan using v2.11.6 or v2.11.7 and the spectrum scan signals are missing with a blank display.  No sure if the problem is in xdr-gtk or with the TEF6686 updated firmware, it (all scan signals in v1.2) works in v2.11.5 with no problems.

PE5PVB, 2024-11-07:
Should be fixed. Please verify

lvxc88, 2024-11-07:
Two questions were found during the test: 1. The previous bandwidth setting will be changed after the scan is completed. 2. If the current band is not FM or OIRT, it seems that there is no available content for the scan. If this function is only used for FM or OIRT, should it be restricted or switched bands before proceeding?

kb7wox, 2024-11-07:
Fixed and working good,

closed 2024-11-20 by PE5PVB