PE5PVB/TEF6686_ESP32 #679: compiling error

lxsxl, 2024-12-10:
Can you please hopefully give me some advise to get beyond the error? Thanks alot!

D:\...\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void setup()':
D:\...\TEF6686_ESP32\TEF6686_ESP32.ino:761:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'
     tft.calibrateTouch(TouchCalData, PrimaryColor, BackgroundColor, 30);
         ^~~~~~~~~~~~~~
D:\...\TEF6686_ESP32\TEF6686_ESP32.ino:781:7: error: 'class TFT_eSPI' has no member named 'setTouch'
   tft.setTouch(TouchCalData);
       ^~~~~~~~
D:\...\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void loop()':
D:\...\TEF6686_ESP32\TEF6686_ESP32.ino:908:13: error: 'class TFT_eSPI' has no member named 'getTouchRawZ'
     if (tft.getTouchRawZ() > 100) {
             ^~~~~~~~~~~~
D:\...\TEF6686_ESP32\TEF6686_ESP32.ino:910:11: error: 'class TFT_eSPI' has no member named 'getTouch'; did you mean 'getColor'?
       tft.getTouch(&x, &y);
           ^~~~~~~~
           getColor

exit status 1

Compilation error: 'class TFT_eSPI' has no member named 'calibrateTouch'

PE5PVB, 2024-12-10:
Did you update the tft\_espi configuration file?

lxsxl, 2024-12-10:
I did so (TFT_eSPI_DynamicSpeed-master.zip but not the old TFT_eSPI-master.zip, correct?).
I re-added the zip-library in arduino IDE but the files in \Documents\Arduino\libraries\TFT_eSPI seem not to be updated (remain on date 2024-11-07).

PE5PVB, 2024-12-10:
Yes, but also edit the User_Setup.h file.

Add:
`//#define ARS                 // uncomment for BGR type display (ARS version)

#define ILI9341_DRIVER
#define TFT_CS          5
#define TFT_DC          17
#define TFT_RST         16
#define TOUCH_CS	32

#define SPI_FREQUENCY       7500000
#define SPI_READ_FREQUENCY 20000000
#define SPI_TOUCH_FREQUENCY 2500000
#define SMOOTH_FONT

#ifdef ARS
#define TFT_RGB_ORDER TFT_RGB
#endif
`

lxsxl, 2024-12-10:
That is exactly included in my User_Setup.h (renamed to .txt):

[User_Setup.txt](https://github.com/user-attachments/files/18086494/User_Setup.txt)

PE5PVB, 2024-12-10:
Are you also sure you didn't update the board manager to v3.x?

lxsxl, 2024-12-10:
[image]

I did this once and stepped back to 1.8.6

lxsxl, 2024-12-10:
[image]

Sorry, that ist the right one...

PE5PVB, 2024-12-10:
That should work. I have no problems compiling. Tested on 2 systems. Maybe another library is fooling you

lxsxl, 2024-12-15:
Proble is resolved, seems that the old TFT_eSPI library was fooling.

lxsxl, 2026-02-21:
I'm sorry, I experience the next compilation error. ChatGPT says it is due to problems with V 3.2.0 but heading back to 2.0.17 leads to a different error. So I hope you have advice for me...

n file included from D:\...\...\TEF6686_ESP32\src\WiFiConnect.h:23, from D:\...\...\TEF6686_ESP32\src\gui.h:7, from D:\...\...\TEF6686_ESP32\src\gui.cpp:1: C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:222:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'? 222 | typedef std::function<String(FS &fs, const String &fName)> ETagFunction; | ^~ | fs::FS In file included from c:\Users\...\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:138, from c:\Users\...\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:101, from D:\...\...\TEF6686_ESP32\src\gui.h:5: C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\FS\src/FS.h:87:7: note: 'fs::FS' declared here 87 | class FS { | ^~ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:222:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'? 222 | typedef std::function<String(FS &fs, const String &fName)> ETagFunction; | ^~ | fs::FS C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\FS\src/FS.h:87:7: note: 'fs::FS' declared here 87 | class FS { | ^~ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:222:60: error: template argument 1 is invalid 222 | typedef std::function<String(FS &fs, const String &fName)> ETagFunction; | ^ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:222:16: error: '<expression error>' in namespace 'std' does not name a type 222 | typedef std::function<String(FS &fs, const String &fName)> ETagFunction; | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:223:32: error: 'ETagFunction' has not been declared 223 | void enableETag(bool enable, ETagFunction fn = nullptr); | ^~~~~~~~~~~~ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:240:3: error: 'ETagFunction' does not name a type; did you mean 'TaskFunction_t'? 240 | ETagFunction _eTagFunction = nullptr; | ^~~~~~~~~~~~ | TaskFunction_t C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:223:50: error: could not convert 'nullptr' from 'std::nullptr_t' to 'int' 223 | void enableETag(bool enable, ETagFunction fn = nullptr); | ^~~~~~~ | | | std::nullptr_t In file included from D:\Lars\Rundfunk\TEF6686_ESP32\src\WiFiConnect.h:23, from D:\...\...\TEF6686_ESP32\src\comms.h:10, from D:\...\...\TEF6686_ESP32\src\comms.cpp:1: C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:222:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'? 222 | typedef std::function<String(FS &fs, const String &fName)> ETagFunction; | ^~ | fs::FS In file included from c:\Users\...\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:138, from c:\Users\...\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:101, from D:\...\...\TEF6686_ESP32\src\comms.h:5: C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\FS\src/FS.h:87:7: note: 'fs::FS' declared here 87 | class FS { | ^~ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:222:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'? 222 | typedef std::function<String(FS &fs, const String &fName)> ETagFunction; | ^~ | fs::FS C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\FS\src/FS.h:87:7: note: 'fs::FS' declared here 87 | class FS { | ^~ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:222:60: error: template argument 1 is invalid 222 | typedef std::function<String(FS &fs, const String &fName)> ETagFunction; | ^ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:222:16: error: '<expression error>' in namespace 'std' does not name a type 222 | typedef std::function<String(FS &fs, const String &fName)> ETagFunction; | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:223:32: error: 'ETagFunction' has not been declared 223 | void enableETag(bool enable, ETagFunction fn = nullptr); | ^~~~~~~~~~~~ C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:240:3: error: 'ETagFunction' does not name a type; did you mean 'TaskFunction_t'? 240 | ETagFunction _eTagFunction = nullptr; | ^~~~~~~~~~~~ | TaskFunction_t C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WebServer\src/WebServer.h:223:50: error: could not convert 'nullptr' from 'std::nullptr_t' to 'int' 223 | void enableETag(bool enable, ETagFunction fn = nullptr); | ^~~~~~~ | | | std::nullptr_t exit status 1 Compilation error: exit status 1