PE5PVB/TEF6686_ESP32 #683: When using the AF function the sound began to stutter a lot

Yurij20, 2024-12-17:
When using the AF function, when AF sheet scanning was activated, and the reception weakened, the sound began to stutter heavily.  This had never happened before and the AF operation was less noticeable by ear.

PE5PVB, 2024-12-17:
That's normal. During stuttering the radio is checking the AF's for signalstrength and PI lock.

PE5PVB, 2024-12-18:
What about 2.11.9?

Yurij20, 2024-12-18:
> What about 2.11.9?

I checked this one too, everything is fine)

PE5PVB, 2024-12-19:
This has to do with speed improvement. I made a change to hopefully undo this on AF scan in https://github.com/PE5PVB/TEF6686_ESP32/commit/74b7bb979e6f19865c14cf3cdf3cba9555eae6eb

Yurij20, 2024-12-20:
> This has to do with speed improvement. I made a change to hopefully undo this on AF scan in [74b7bb9](https://github.com/PE5PVB/TEF6686_ESP32/commit/74b7bb979e6f19865c14cf3cdf3cba9555eae6eb)

I checked.  I made a second comparison with firmware 2.11.9 and the current 2.11.11.  It has gotten better, but the search process by af still periodically interrupts the sound.  For example, three clicks occur quietly, and the fourth interrupts the sound as before.  Yes, and quiet clicks also seem to be audible a little more than on 2.11.9

lxsxl, 2024-12-20:
With the latest code there is no AF tuning at all. There are the typical dropouts (tuning attempts?) but it never switches to an AF even if there are strong signals on several decoded AFs.

PE5PVB, 2024-12-24:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/e27c2f504e4bf13b521b0988c3ef01aa03e0873c

Yurij20, 2024-12-25:
> Fixed in [e27c2f5](https://github.com/PE5PVB/TEF6686_ESP32/commit/e27c2f504e4bf13b521b0988c3ef01aa03e0873c)

Fixed, thanks:)

Yurij20, 2025-04-13:
2.20.  Auto scanning AF list occurs periodically with short -term hiss.   Narimer three clicks occur dullly as if it is, the fourth with hissing.  Sometimes everyone hisses.  I compared with the early firmware there everything is clear clicks of the search as deaf as it should be

closed 2025-06-06 by PE5PVB