PE5PVB/TEF6686_ESP32 #690: 2.11.13: Config wifi connection

numerci, 2024-12-24:
1. Go to the config wifi connection
Result:
a) the texts are overlapped with the new touch options
b) that menu is a trap... I cannot close it in any way. Just by switching off the radio the user can exit.

PE5PVB, 2024-12-24:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/e87fd97b759dd963aef066c4a64e0a60c8dd5c0b

numerci, 2025-01-02:
Well... partially fixed.
a) fixed and there is no overlapping text
b) despite the new message stating to press Mode to exit the screen when the Mode button is pressed nothing happens. The menu is still a trap in 2.11.14

PE5PVB, 2025-01-02:
Yeah, it's not possible to exit wifi config. Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/8e98af5a705398af340fa512e8c938bdea338c63

numerci, 2025-01-07:
Unfortunately, there is no change for this in 2.11.15. Still no way to close the message unless the radio is switched off.

PE5PVB, 2025-01-07:
Again. You can't escape the WiFi configuration. Only after a succesfull configuration OR a reboot. This is just how the current library works.

numerci, 2025-01-07:
Ah, sorry I have misunderstood that it was fixed as there was a change and the fix commit. Closed.

closed 2025-01-07 by numerci