PE5PVB/TEF6686_ESP32 #703: 2.11.15: Last letter from the PS stored name disappears for a second

numerci, 2025-01-07:
1. Memorize a station that has a long PS (7-8 chars).
2. Zap between memorized stations
3. Check how the PS is displayed.
Result: Most of the times the last char from the PS is missing for a second or two. Then is displayed.

PE5PVB, 2025-01-08:
Is this a new bug appeared in the last update?

numerci, 2025-01-08:
Yes, it appeared on 2.11.15. With 2.11.14 is working fine. Apparently there is a refresh of the PS that was added with the latest version that generates this issue.

PE5PVB, 2025-01-10:
I was not able to reproduce this. Maybe those channels have low signal quality where RDS has some errors?

numerci, 2025-01-10:
No, it does this for all stations no matter the signal level. They just need to have a longer name in PS

numerci, 2025-01-10:
Here it is: https://youtu.be/KUxGRAovL0g

PE5PVB, 2025-01-10:
I see what happens. Let me try something. Please test the next beta

numerci, 2025-01-10:
Works fine on 2.11.17. Closed

closed 2025-01-10 by numerci