PE5PVB/TEF6686_ESP32 #706: RDS clock is not displayed in the Main screen

numerci, 2025-01-08:
When the Show Clock is enabled and tuning on a station that broadcasts the time the hour is not displayed on the Main Screen anymore.

Note: this was noticed on 2.11.16 but apparently is not working on the 2.11.15 either. Usually I keep this option off because the time for most of the stations is wrong so I do not know exactly when it stopped working.

PE5PVB, 2025-01-09:
I was not able to reproduce. Can you make a video?

numerci, 2025-01-10:
Here it goes: https://youtu.be/7EvLlotB_hs

Same station in 3 different versions starting from the bottom: 2.11 , middle 2.11.16 and top 2.11.14

PE5PVB, 2025-01-10:
Wait somewhat longer. Will it display? I have no problems with the current firmware. Maybe it can take somewhat longer in the current firmware, because of the way RDS is processed.

numerci, 2025-01-10:
Indeed ,it looks like is loading but it takes very very long time. I took minutes. in 2.11 is loading before the screen goes off (set to 60 sec)

PE5PVB, 2025-01-10:
I know. This is not a bug, but is the result of a cleaner RDS processing. CT isn't transmitted that often in RDS, so you just have to wait somewhat longer.

closed 2025-01-10 by numerci