PE5PVB/TEF6686_ESP32 #714: Does not display radio text on our radio station

Yurij20, 2025-01-11:
I have long noticed this problem with one of the radio stations, but now I decided to report it. There is no display of radio text from this radio station, although in fact it is present on it. I am attaching the MPX recording, everything is clearly visible from it
[image]

https://disk.yandex.ru/d/N4CnPZTTVntVwA

PE5PVB, 2025-01-12:
I tried to use this MPX recording, but it's unusable. Please send a new track and test it before sending using RDS Spy.

SpiderBlocker, 2025-01-12:
I have normalized this wav-file and while the pilot level is correct, the RDS-level is extremely low. It needs to be increased by at least 10 dB or so. Also there's an ugly spike at 76 kHz. This radio station needs some serious adjustments ...
[image]

Yurij20, 2025-01-13:
> I tried to use this MPX recording, but it's unusable. Please send a new track and test it before sending using RDS Spy.

Apparently I recorded it at a low level, try this file, if it doesn't work I'll try to re-record it in the evening
https://disk.yandex.ru/d/kBI7VWfrBucEaw

SpiderBlocker, 2025-01-13:
Hi Yurij20, haven't you seen my reply above? Your station's RDS injection level is way too low, that is the problem why radio's will have trouble decoding it. Please make sure that the RDS injection level is at 4,5%.

SpiderBlocker, 2025-01-13:
I normalized your new file as well and verified with MPX analyzer. The RDS-level appears to be around 2%. Also the average noise floor is very high, and as such a receiver will have additional challenge to properly decode it. My analyzer only managed to get the  PS "Zvezda" and "95.6 MHz". 
Now Zvezda appears to be a Russian FM army and in your supplied wav-file they were discussing about how they attacked an energy facility of the military-industrial complex of Ukraine. I am not really sure whether you are in support of that radio station or not. If so, I think it is rather inappropriate asking in this forum for assistance to help such station to improve their reception in any way.

PE5PVB, 2025-01-13:
OK. I can use the file after normalisation. I see that the station uses RT over block B. I thought it should work but was never able to test it, because all radiostations nearby are using block A. I'll fix this issue this week.

Yurij20, 2025-01-13:
> I normalized your new file as well and verified with MPX analyzer. The RDS-level appears to be around 2%. Also the average noise floor is very high, and as such a receiver will have additional challenge to properly decode it. My analyzer only managed to get the PS "Zvezda" and "95.6 MHz". Now Zvezda appears to be a Russian FM army radio station and in your supplied wav-file they were discussing about how they attacked an energy facility of the military-industrial complex of Ukraine. I am not really sure whether you are in support of that radio station or not. If so, I think it is rather inappropriate asking in this forum for assistance to help such station to improve their reception in any way.

As for politics, I’m not interested in this topic at all, I discovered this radio station that there was an error with the radio station, that’s why I contacted it and what was transmitted on it is not interesting to me

andimik, 2025-01-13:
Just a question: Does Rdsspy (or redsea) show RT of this radio station?

PE5PVB, 2025-01-13:
> Just a question: Does Rdsspy (or redsea) show RT of this radio station?

RDSSpy does work ok.

Yurij20, 2025-01-14:
> Fixed in [33ce573](https://github.com/PE5PVB/TEF6686_ESP32/commit/33ce573c354c17a38d5ad249005f3ae928754a26)

Fixed, thanks :)

closed 2025-01-14 by Yurij20