PE5PVB/TEF6686_ESP32 #737: New ODA register

lxsxl, 2025-01-19:
Last year a new ODA register was published. If you tell me where to find it, I can care for the update but due to a week on holiday it might take a while.

[R24_006_1_ODA register.pdf](https://github.com/user-attachments/files/18470519/R24_006_1_ODA.register.pdf)

PE5PVB, 2025-01-19:
Fleximax?

lxsxl, 2025-01-19:
Should be the only update if you used the register of 2023 until now.

PE5PVB, 2025-01-19:
I'll check

lxsxl, 2025-01-26:
I sent a pull request

lxsxl, 2025-01-26:
Seems that the scrollbar with information on used ODAs got lost. In AF and EON screens there is only "PRESS BAND TO CLOSE".

PE5PVB, 2025-01-26:
Yes I know. Do you know in which beta it was lost?

lxsxl, 2025-01-26:
Not yet but I will test. 20 (before my changes!) didn’t work either.

lxsxl, 2025-01-26:
Up to 2.11.2 worked fine, since 2.11.3 it is faulty,

PE5PVB, 2025-01-28:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/f6c439c57ebfdb0bb91417565c708bdddef4ab65
Please check if you didn't make any mistakes in your AID table update

lxsxl, 2025-01-28:
There is at least one error, because CD46 is displayed as ELECTRABEL-DSM 8 (should be TMC/ALERT-C). But I have a problem with contribution (at least I don't know what to do): My branch ist 2 commits ahead of your main. How do I get it "in shape" again?

PE5PVB, 2025-01-28:
Shall I restore the old data?

lxsxl, 2025-01-28:
No, in row 507 there is a missing comma. It should fit then.

But how do I get my branch back to your state of developement?

PE5PVB, 2025-01-28:
OK. I reverted it. Please add the data and test it first. Then you can merge/sync in into my repository.

lxsxl, 2025-01-28:
Sent a pull request.

closed 2025-01-28 by lxsxl