PE5PVB/TEF6686_ESP32 #745: 2.11.20: When stopping an DX Scan the memory number remains green

numerci, 2025-01-22:
1. Go to Memory channels mode.
2. Start the DX Scan (either from Menu or from the button).
3. Wait for some seconds for the Scan to start.
4. Press the BW button to stop the scan.
Result: The memory channel number is green instead of yellow.

PE5PVB, 2025-01-23:
Tried to reproduce, but was not able to. Are there any memory channels available in your test?

numerci, 2025-01-23:
Looks like there is a precondition for this to happnen. Set the start memory and stop memory outside the memorizes stations. Example I have stations saved from 1 to 28 and I have put start to 30 and end to 33 so it logs every station. In this case the steps above will result in the green memory number.

PE5PVB, 2025-01-23:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/13c17d7fb7512f17a7836da1208cff66f7b485e0

numerci, 2025-01-29:
works fine on 2.11.21

closed 2025-01-29 by numerci