PE5PVB/TEF6686_ESP32 #749: PS color is not separated by ps[xx]error

andimik, 2025-01-25:
In https://github.com/PE5PVB/TEF6686_ESP32/commit/acca9dccf728c42c4ff610f7744228f859dde108 you have introduced separate colors regarding PS quality. The source code seems to have PS separated into PS positions 1+2, 3+4, 5+6, 7+8 (which is my expectation, as PS is transmitted in 4x2 characters).

But in real, the color for all 8 characters is the same, regardless their own position quality.

Meaning: For weaker signals, we cannot distinguish if `__` are really two spaces or just not received/decoded.

PE5PVB, 2025-01-28:
Not sure what you mean. Do you think there is a problem, or did you see that there was a problem?

lxsxl, 2025-01-28:
What I think andimik means is that the PS should be in different colours for each pair of characters: if one pair is received without error it should be in primary colour, other pairs that are received possibly faulty should stay in secondary colour until they are also surely completed.

For beta 2.11.15 you wrote "show errors in PS with other color" but the whole PS is still in primary color (if RDS-stream is synchronized) or (if not) it is in secondary color. But maybe only misinterpretation?

PE5PVB, 2025-01-28:
It works exact as it should. Maybe the code looks a bit weird, that is because it works together with the RDS parser.

lxsxl, 2025-01-28:
For me this issue can be closed

andimik, 2025-01-28:
Well, for example, if I tune to (let's say) a station with PS `RADIO_24` and I see `RADIO___` in yellow I don't know if this is already the full PS or not.

I expect `RADIO_` in yellow and `__` in blue.

PE5PVB, 2025-01-28:
That is because the __ bytes aren't received yet.

andimik, 2025-01-28:
Any why not skip them? `RADIO_  ` in order to distinguish between a space (20) and not decoded yet (NULL)?

PE5PVB, 2025-01-29:
I understand what you want, but this would take me a complete rewrite of a piece of code, which I currently don't want to do.