PE5PVB/TEF6686_ESP32 #752: PTY enhancement

lxsxl, 2025-01-26:
Would you agree if we add the PTY number to the textstrings (e. g. "10 Pop Music" oder "Pop Music 10" instead of only "Pop Music"? This would help to better learn and remember the content if only the number is avalable (like on EON screen or in logbook file). I am testing this already for a while now with my own code. If you agree I would update the language table.

lxsxl, 2025-01-26:
Forgot to mention: At least with the german translations there is no overlapping with the time-field.

PE5PVB, 2025-01-28:
Added in https://github.com/PE5PVB/TEF6686_ESP32/commit/08fced7eeccd270104b6262c734ac06ebdc2d60e

lxsxl, 2025-01-28:
Thanks, it works fine. I'll close this issue.

closed 2025-01-28 by lxsxl

radiola2, 2025-04-11:
I think it should go to an "Advanced options" choice in the menu as I think the number is an annoyance.