PE5PVB/TEF6686_ESP32 #757: Switching wifi on lets reboot TEF

andimik, 2025-01-28:
when you set wifi active in the menu, it will reboot TEF without chance to activate it.

Tested with 2.11.21 (+ fix for Highpoint)

andimik, 2025-01-28:
Workaround: reconfigure Wifi again.

numerci, 2025-01-29:
I can confirm that this happens to me also. If I enable wifi the TEF reboots instantly. Re configuring the wifi sorts out the issue.

PE5PVB, 2025-01-29:
Did this happen once, or can you reproduce it time over time?

numerci, 2025-01-29:
It is always reproducible after update from 2.11.20. Once the wifi is re-configured (workarround above) it never occurs. Looks like some incompatible setting coming from the previous version. Once that is removed the rebooting doesn't occur.

PE5PVB, 2025-01-29:
OK. Well, in that case it is not a bug. I think something internal in ESP32 core. Well, once it will come to a public release, the complete flash will be resetted anyway.

numerci, 2025-01-29:
Are you sure about resetting everything? I though that at least the memorized stations should remain. But on the other side is more safe to erase everything...

PE5PVB, 2025-01-29:
Well, assume you switch from v2.11 to v2.20 a lot of new variables aren't set and can cause some weird behaviour.

andimik, 2025-01-29:
I've updated each time you have published a new beta but this is the first time that I saw this behavior.

Overland-DX, 2025-01-30:
I can confirm the same problem with WIFI. When I set up the WIFI settings again, it worked. but when I went to re-enable WIFI after a few hours, the radio rebooted again.

numerci, 2025-01-30:
Indeed today is not working again although after setting it again yesterday it was working fine.

andimik, 2025-01-30:
> the complete flash will be resetted anyway

Including Wifi-Settings? If not, you should reset them as well please, otherwise the issue will remain for the final version.

PE5PVB, 2025-01-30:
Thanks for your input. Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/b73f0fe3f608546bd274d0e4e4cfd8cc6fcf24d4

Overland-DX, 2025-01-30:
> Thanks for your input. Fixed in [b73f0fe](https://github.com/PE5PVB/TEF6686_ESP32/commit/b73f0fe3f608546bd274d0e4e4cfd8cc6fcf24d4)

Confirming that WIFI works fine here now.

closed 2025-01-31 by andimik