PE5PVB/TEF6686_ESP32 #765: "New" clock

lxsxl, 2025-01-31:
Some issues with the new clock position:

1. If you make an AF-scan the clock will disappear.
2. If a clock was received and you tune to a different QRG there will be an overlay (in my case 12 h-mode over 24 h-mode, see pic).
3. Now that you found a different place for the clock it would be great to add date and weekday, at least a short thing like "Fr, 31.01.2025" (likely also a setting for format is needed) to the screen. For that you could reduce the space of stereo and squelch like you did for the time. Date was a wish for a long time I never suggested as I always thougt we were "full"...

[image]

PE5PVB, 2025-01-31:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/6c39f4840e901ed29dac7787932431c22dd238d7

Where on earth would you like to add the date?

lxsxl, 2025-01-31:
As you reduced the space for the aviation (or what is it called below the clock) you could do the same with stereo and squelch labels.

PE5PVB, 2025-01-31:
I'll discuss with my "graphical designer"

lxsxl, 2025-01-31:
:-)

lxsxl, 2025-02-01:
The “ghost clock” has disappeared now, so that’s fine. But the clock is still lost if you switch to AF-mode and also if you switch from AF-mode to EON-mode (or back).

PE5PVB, 2025-02-04:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/07d33a648753ab6ea0b20bdd58a77ac16b2bcae9

lxsxl, 2025-02-04:
Fix was successful, thanks!
Shall we keep this issue due to the "date to be implemented"?

PE5PVB, 2025-02-04:
We have to see if we can squeeze it in somewhere without having the GUI look lame when no CT is available. Also, date isn't that important.

lxsxl, 2025-02-07:
Wow, you made it :-) 
It looks great and works fine. I have two further ideas, unfortunately also space requiring, but I will place them as ideas.
For now I close this issue.

closed 2025-02-07 by lxsxl