PE5PVB/TEF6686_ESP32 #767: PTY in EON view shows nothing instead of 0

andimik, 2025-02-01:
All my RDS EON programs use EON-PTY, but declare it as `No PTY`, regardless the real PTY.

```
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA203","tp":false,"prog_type":"No PTY","ta":false}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA213","tp":false,"ps":"  FM4   "}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA213","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA902","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA201","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA213","tp":false,"ps":"  FM4   "}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA213","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA902","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA201","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA203","tp":false,"prog_type":"No PTY","ta":false}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA213","tp":false,"ps":"  FM4   "}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA902","tp":false,"ps":"RADIO-ST"}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA902","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA201","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA203","tp":false,"ps":"OE 3    "}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA203","tp":false,"prog_type":"No PTY","ta":false}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA213","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA902","tp":false,"ps":"RADIO-ST"}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA902","tp":false,"prog_type":"No PTY","ta":true}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA201","tp":false,"ps":"  OE 1  "}}
{"pi":"0xA502","group":"14A","tp":true,"prog_type":"No PTY","other_network":{"pi":"0xA201","tp":false,"prog_type":"No PTY","ta":true}}
```

My expectation is that the last column shows `0` for `No PTY`, but it's empty.

[image]

Tested with 2.11.21 from Discord.

Do you need to set the variable to 254?

andimik, 2025-02-01:
And why not use `O` for `yes` and `X` for a definitely `no`.  Because also here we cannot distinguish if the value is `no` or just not received yet.

andimik, 2025-02-01:
I just found a program where it is PTY other than 0, here it works.

```
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5203","tp":true,"ps":"RADIO3  "}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5203","tp":true,"has_linkage":false}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5203","tp":true,"prog_type":"Culture","ta":false}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5202","tp":true,"ps":"RADIO2  "}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5202","tp":true,"has_linkage":false}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5202","tp":true,"prog_type":"Easy listening","ta":false}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5203","tp":true,"ps":"RADIO3  "}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5203","tp":true,"has_linkage":false}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5203","tp":true,"prog_type":"Culture","ta":false}}
{"pi":"0x5201","group":"14A","tp":true,"prog_type":"News","other_network":{"pi":"0x5202","tp":true,"ps":"RADIO2  "}}
```

[image]

lxsxl, 2025-02-07:
I would agree to andimiks suggestion: keep PTY empty if PTY is not transmitted (no variant 13 in group 14A) but show a "0" if hex 00000 is transmitted in variant 13 (see figure from RDS standard).

[image]

andimik, 2025-02-07:
Yes, we already had similar discussions for ECC and LIC values, where 0 is a value and not NULL.

PE5PVB, 2025-02-11:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/a99eba34001833d2d64aa1a6dff3adc1f4151f42

andimik, 2025-02-12:
Yes, confirmed, too.