PE5PVB/TEF6686_ESP32 #777: Compilation error: 'class TFT_eSPI' has no member named 'setSPISpeed'

SKOCHE, 2025-02-07:
Hi. Encountered a compilation error: 'class TFT_eSPI' has no member named 'setSPISpeed'. Tell me how you can solve it.

C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void setup()':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:605:9: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     tft.setSPISpeed(SPI_FREQUENCY / 1000000);
         ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:609:9: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     tft.setSPISpeed(spispeed * 10);
         ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:806:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'
     tft.calibrateTouch(TouchCalData, PrimaryColor, BackgroundColor, 30);
         ^~~~~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:826:7: error: 'class TFT_eSPI' has no member named 'setTouch'
   tft.setTouch(TouchCalData);
       ^~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void loop()':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:964:13: error: 'class TFT_eSPI' has no member named 'getTouchRawZ'
     if (tft.getTouchRawZ() > 100) {  // Check if the touch is active
             ^~~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:966:11: error: 'class TFT_eSPI' has no member named 'getTouch'; did you mean 'getColor'?
       tft.getTouch(&x, &y);
           ^~~~~~~~
           getColor
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void ModeButtonPress()':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:2575:36: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
             if (spispeed == 7) tft.setSPISpeed(40);
                                    ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void DoMemoryPosTune()':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:3056:26: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
   if (spispeed == 7) tft.setSPISpeed(50);
                          ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void setAutoSpeedSPI()':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5046:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 875 ... 877: tft.setSPISpeed(28); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5047:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 878 ... 881: tft.setSPISpeed(24); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5048:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 882 ... 892: tft.setSPISpeed(42); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5049:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 893 ... 899: tft.setSPISpeed(31); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5050:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 900 ... 904: tft.setSPISpeed(12); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5051:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 905 ... 906: tft.setSPISpeed(16); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5052:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 907 ... 910: tft.setSPISpeed(11); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5053:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 911 ... 916: tft.setSPISpeed(15); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5054:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 917 ... 921: tft.setSPISpeed(24); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5055:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 922 ... 928: tft.setSPISpeed(13); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5056:19: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 929: tft.setSPISpeed(11); break;
                   ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5057:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 930 ... 932: tft.setSPISpeed(13); break;
                           ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5058:27: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 933 ... 939: tft.setSPISpeed(18); break;
[... 75 lines cut ...]
                             ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5084:29: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 1069 ... 1074: tft.setSPISpeed(14); break;
                             ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5085:20: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 1075: tft.setSPISpeed(17); break;
                    ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5086:29: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     case 1076 ... 1080: tft.setSPISpeed(15); break;
                             ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5087:18: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
     default: tft.setSPISpeed(30); break;
                  ^~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void NumpadProcess(int)':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:5427:30: error: 'class TFT_eSPI' has no member named 'setSPISpeed'
       if (spispeed == 7) tft.setSPISpeed(40);
                              ^~~~~~~~~~~

exit status 1

Compilation error: 'class TFT_eSPI' has no member named 'setSPISpeed'

[image]

[image]

PE5PVB, 2025-02-07:
Looks like you didn't use the modified tft_espi library.

SKOCHE, 2025-02-07:
> Looks like you didn't use the modified tft_espi library.
Can you give me a link to this library?

PE5PVB, 2025-02-07:
https://github.com/ohmytime/TFT_eSPI_DynamicSpeed/archive/refs/heads/master.zip

SKOCHE, 2025-02-07:
> https://github.com/ohmytime/TFT_eSPI_DynamicSpeed/archive/refs/heads/master.zip

Thank you. Now there is such a mistake.
In file included from C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:11:
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\WebServer\src/WebServer.h:222:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'?
  222 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                                ^~
      |                                fs::FS
In file included from C:\Users\SK\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:138,
                 from C:\Users\SK\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:101,
                 from C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:9:
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\FS\src/FS.h:87:7: note: 'fs::FS' declared here
   87 | class FS {
      |       ^~
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\WebServer\src/WebServer.h:222:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'?
  222 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                                ^~
      |                                fs::FS
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\FS\src/FS.h:87:7: note: 'fs::FS' declared here
   87 | class FS {
      |       ^~
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\WebServer\src/WebServer.h:222:60: error: template argument 1 is invalid
  222 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                                                            ^
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\WebServer\src/WebServer.h:223:50: error: could not convert 'nullptr' from 'std::nullptr_t' to 'WebServer::ETagFunction' {aka 'int'}
  223 |   void enableETag(bool enable, ETagFunction fn = nullptr);
      |                                                  ^~~~~~~
      |                                                  |
      |                                                  std::nullptr_t
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\WebServer\src/WebServer.h:240:32: error: cannot convert 'std::nullptr_t' to 'WebServer::ETagFunction' {aka 'int'} in initialization
  240 |   ETagFunction _eTagFunction = nullptr;
      |                                ^~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void setup()':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:806:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'
  806 |     tft.calibrateTouch(TouchCalData, PrimaryColor, BackgroundColor, 30);
      |         ^~~~~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:826:7: error: 'class TFT_eSPI' has no member named 'setTouch'
  826 |   tft.setTouch(TouchCalData);
      |       ^~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void loop()':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:964:13: error: 'class TFT_eSPI' has no member named 'getTouchRawZ'
  964 |     if (tft.getTouchRawZ() > 100) {  // Check if the touch is active
      |             ^~~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:966:11: error: 'class TFT_eSPI' has no member named 'getTouch'
  966 |       tft.getTouch(&x, &y);
      |           ^~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void ScreensaverTimerInit()':
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:1485:30: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
 1485 |   timScreensaver = timerBegin(0, 80, true);
      |                    ~~~~~~~~~~^~~~~~~~~~~~~
In file included from C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal.h:98,
                 from C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/Arduino.h:36,
                 from C:\Users\SK\AppData\Local\arduino\sketches\9787B8CC6E2E08C784618C03FFC1D506\sketch\TEF6686_ESP32.ino.cpp:1:
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:1486:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
 1486 |   timerAttachInterrupt(timScreensaver, ScreensaverInterrupt, true);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\SK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino: In function 'void ScreensaverTimerSet(byte)':
[... cut ...] n 'timerWrite'?
 1499 |     timerAlarmWrite(timScreensaver, value * TIMER_SCREENSAVER_BASE, true);
      |     ^~~~~~~~~~~~~~~
      |     timerWrite
C:\Users\SK\Desktop\TEF6686_ESP32-main\TEF6686_ESP32\TEF6686_ESP32.ino:1501:5: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
 1501 |     timerAlarmEnable(timScreensaver);
      |     ^~~~~~~~~~~~~~~~
      |     timerAlarm

exit status 1

Compilation error: 'class TFT_eSPI' has no member named 'calibrateTouch'

PE5PVB, 2025-02-07:
Please use ESP32 boardmanager v2.0.17. Looks like you are using v3.x

closed 2025-02-07 by PE5PVB