PE5PVB/TEF6686_ESP32 #790: Canada PI Code database support?

pclov3r, 2025-02-11:
It would be nice to see a Canada pi code database added. This in the past has been added to SDR-Console so perhaps that could be used as a refernece.

https://ised-isde.canada.ca/site/spectrum-management-telecommunications/en/licences-and-certificates/broadcasting/program-information-codes-radio-broadcasting-stations explains how PI codes work and are allocated for Canada.

PE5PVB, 2025-02-25:
Added in https://github.com/PE5PVB/TEF6686_ESP32/commit/423de6d3f154f166ea8344785d3a8f5e1150da7e
For manual callsigns which can't be obtained from the calculator there is a database you can fill. It's in TEF6686.h

lxsxl, 2025-02-26:
Now that we have the new region "Canada" wouldn't it be correct to change "America" to "USA"?

lxsxl, 2025-02-28:
My last comment is related to item 48 in the language file

pclov3r, 2025-02-28:
> Now that we have the new region "Canada" wouldn't it be correct to change "America" to "USA"?

Perhaps `America` Should be changed to `North America` now?

lxsxl, 2025-02-28:
Actually we have "Europe", "America" and "Canada". For me it doesn't make sense. And "North America" instead of "America" wouldn't make it better. But probably there is a reason that we keep it like it is? I'm not so familiar with the situation in America...

pclov3r, 2025-03-03:
> Actually we have "Europe", "America" and "Canada". For me it doesn't make sense. And "North America" instead of "America" wouldn't make it better. But probably there is a reason that we keep it like it is? I'm not so familiar with the RDBS-situation in America...

Hum... Perhaps that discussion needs to be moved to a new issue.

For now i'll close this as it's been implemented in https://github.com/PE5PVB/TEF6686_ESP32/commit/423de6d3f154f166ea8344785d3a8f5e1150da7e

closed 2025-03-03 by pclov3r