PE5PVB/TEF6686_ESP32 #800: 2.11.24RC: The M and S bars are not easy visible on the Whiteout theme

numerci, 2025-02-25:
2.11.24RC: The M and S bars are not easy visible on the Whiteout theme

MCelliotG, 2025-02-26:
I will add the Green color back, however a Whiteout theme should not have colors. There are not many options here.

MCelliotG, 2025-02-26:
> In the standard theme, the middle region color is too dark (brownish/orange) as opposed to the left (bright green) and right (bright red).

Color theory, green and red make brown, so the gradient that mixes these colors in the middle is brown. An option here would be to add a third color in the middle, but it has to be done in coding.

SpiderBlocker, 2025-02-26:
> Color theory, green and red make brown, so the gradient that mixes these colors in the middle is brown. An option here would be to add a third color in the middle, but it has to be done in coding.

I would suggest to interpolate in HSV rather than in RGB, which should result in a far more bright mixed region (yellow, rather than the current brownish), since the brightness (V) would remain constant.

PE5PVB, 2025-02-27:
> I would suggest to interpolate in HSV rather than in RGB, which should result in a far more bright mixed region (yellow, rather than the current brownish), since the brightness (V) would remain [...]

Thanks for the tip. See https://github.com/PE5PVB/TEF6686_ESP32/commit/160bee883f9ec141d199891a3a3d3d8d339c5ca8

SpiderBlocker, 2025-02-27:
Great, thx!

One more little question, it seems that the S-meter won't fill up competely (not sure about the mod-meter), not even at a level beyond 90 dBµV in the FM band, where the receiver's input actually starts to saturate:

[image]

PE5PVB, 2025-02-27:
> I can't manage to see the RDS clock no more, whereas it is enabled.
Did you check in connectivity menu if CT locked to PI is disabled? Maybe it's enabled to a wrong PI?

SpiderBlocker, 2025-02-27:
> > I can't manage to see the RDS clock no more, whereas it is enabled.
> > Did you check in connectivity menu if CT locked to PI is disabled? Maybe it's enabled to a wrong PI?

Yep, was locked to PI FFFF for some unknown reason, all good thx!

PE5PVB, 2025-02-28:
Full meter problem fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/769252df7ed7d689e6d15bd96ef5f79773b70e59

closed 2025-02-28 by numerci