PE5PVB/TEF6686_ESP32 #801: 2.11.24:RC: The M and S bars have the colors reversed on some of the Themes

numerci, 2025-02-25:
The M and S bars have the colors reversed on some of the Themes:
1. Tangerine
2. Ocean
3. Indigo

MCelliotG, 2025-02-26:
They are not reversed, they start with different colors. There's no rule stating that they should both be the same gradient. However I can make them end on the same color if certain users feel they'll look better.

numerci, 2025-02-26:
I am not using those teams so if you say that this is how they should be is fine with me. The bug was added because a similar bug existed and it was fixed for the main theme. If for the other themes this is fine then the bug can be closed.

MCelliotG, 2025-02-26:
It's not a problem, I've already changed the ending color to be the same on these themes and pull a request in a moment...

PE5PVB, 2025-02-28:
Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/d1c194682586ccb4a8eefbd63572570c63a91271

MCelliotG, 2025-02-28:
_"Someone who knows how these should actually look"_ you're funny! These were done in purpose for the Queer theme. I might change just the Volcano theme.

closed 2025-02-28 by numerci