PE5PVB/TEF6686_ESP32 #807: When 5 second long bandpass button is enabled, the radio doesn't fully shut down.

pclov3r, 2025-03-03:
When 5 second long bandpass button is enabled, The radio doesn't fully shut down. The LED light stays illuminated. 

This is at least on the DP-666. 

Not sure if other models behave the same way.

PE5PVB, 2025-03-03:
It will put the display, ESP and TEF into standby, but not the other hardware. This is a hardware problem, not software.

pclov3r, 2025-03-03:
Any opinions of maybe re-naming it to standby to make it a bit more clear despite hardware issues?

PE5PVB, 2025-03-04:
The firmware is made for all kinds of hardware designs. For example, the original DIY kit really does go to stand-by with this function. So I see no need to change this.

pclov3r, 2025-03-05:
> The firmware is made for all kinds of hardware designs. For example, the original DIY kit really does go to stand-by with this function. So I see no need to change this.

Thanks!

Since this is intended I'll go ahead and close this too.

closed 2025-03-05 by pclov3r