PE5PVB/TEF6686_ESP32 #812: PI-Code 0000

lxsxl, 2025-03-11:
I just remembered one issue I recognized some months ago and I’m not sure if this was reported (and possibly solved): If PI code is “0000” (of course this is not according to the RDS standard, but it’s real) this PI is not shown. It would be better to show even the not appropriate “0000” to see a difference to “not detected”. If needed I can make a recording but it may take some days or even weeks until I get to one of those transmitters.

andimik, 2025-03-11:
Hint: In FMLIST there were lots of logs from Caribbean sea and Gulf of Mexico with some stations transmitting 0x0000 as PI-Code, indeed.

So it's quite common in countries where there is no or only limited regulation.

PE5PVB, 2025-03-11:
Problem is that when I show PI code 0000 it often just pops up when listening to static, that's why 0000 is not shown. 0000 is the same as no PI code.

andimik, 2025-03-11:
How does the RDS groups look like in such a case? Is group A skipped like in `---- bbbb cccc dddd`?

PE5PVB, 2025-03-11:
No it is filled with 0000 ofcourse.

lxsxl, 2025-03-11:
The TEF displays only spaces (not 0000!) as PI as long as no RDS stream is received. So it should be possible to show "0000" if this is transmitted in blocks A.

lxsxl, 2025-03-12:
The rest will be displayed even if PI field remains empty. See 1:27:40 in this video:
https://m.youtube.com/watch?v=4fpjPwSAE1Y&pp=ygUTQmFuZHNjYW4gTHV4ZW1idXJnIA%3D%3D
This one of the two stations in Germany with PI 0000.

WIHDIA, 2025-03-14:
Just checked v2.11.27 and no PI 0000 on 101.7 AFN Everberg (BEL).

[image]

andimik, 2025-03-14:
Maybe you share an mpx recording with 192 kHz?

WIHDIA, 2025-03-14:
Is a recording made with SDRConsole ok? AFN 101.7 has been lacking audio for weeks now. I've been informing them about this about a dozen times and it's still a mess.

andimik, 2025-03-15:
Sjef: I found one on a spyserver in Mexico.

[image]

192 kHz sampling MPX file

[image]

[pi_0000.zip](https://github.com/user-attachments/files/19262767/pi_0000.zip)

andimik, 2025-03-15:
And this is in FLAC, 30 seconds.

[pi_0000.zip](https://github.com/user-attachments/files/19262804/pi_0000.zip)

andimik, 2025-03-15:
Or from Indonesia:

[image]

[image]

[pi_0000_indonesia.zip](https://github.com/user-attachments/files/19262876/pi_0000_indonesia.zip)

[image]

andimik, 2025-03-15:
Even Turkish Public Broadcaster TRT has technicians not knowing what the PI Code stands for ...

[image]

WIHDIA, 2025-03-15:
101.7 AFN Everberg right om my doorstep.

[image]

[image]

[image]

Recording: https://wijnants.stackstorage.com/s/aC7fMlMCjudyWz4d

73
Kind regards
Met vriendelijke groeten

Herman Wijnants

[My logblog](https://www.wijnants.info/fmdx/)
Belgium editor [fmlist.org](https://www.fmlist.org/)

lxsxl, 2025-03-16:
I also made a record. This is from "Radius", the university-radio in Siegen (NRW):

https://wetransfer.com/downloads/7c2c476bab27c17f36bcba14399b05f820250316181236/c33062445f23825270959454568eb48c20250316181305/31d295?t_exp=1742407956&t_lsid=51842295-459d-4bde-a3bb-d8860ac7924d&t_network=email&t_rid=ZW1haWx8Njc3MWNhOWE0ZGZkZTcwNmYwZmI0YWJk&t_s=download_link&t_ts=1742148785&utm_campaign=TRN_TDL_01&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_01

lxsxl, 2025-03-16:
@ Herman: Your file seems not to be readable by RDSSpy, seems that the level is too low. And you can save a lot of memory if you record it at 192 kHz and 16 bit. 2,5 GB for 1 minute is definitely too large ;-)

@ Andi: Is it possible to convert the flac-files so they can be read in RDSSpy? Who would it work?

andimik, 2025-03-16:
Sure, it should be working with `sox` as well.

PE5PVB, 2025-03-16:
Just posted a hotfix for PI 0000

andimik, 2025-03-16:
Sjef, did the files help you?

PE5PVB, 2025-03-16:
Yes, I used one of the posted files

pclov3r, 2025-03-16:
If needed, I can test this with a TX sdr and a RDS encoder with anything in there.

WIHDIA, 2025-03-17:
It's working well on 101.7 AFN Everberg.

[image]

lxsxl, 2025-03-17:
Same thing in Siegen:

[image]

lxsxl, 2025-03-17:
As there is positive feedback from multiple sites I close this issue as completed.
Thanks Sjef!

closed 2025-03-17 by lxsxl