PE5PVB/TEF6686_ESP32 #813: Number buttons don't work

HB9ERD, 2025-03-14:
Hi Sjef
I have got today my new DP-666 and flashed the latest beta firmware.
Unfortunately, the number buttons don't work. Every number button and also the buttons named "DX" and the one with the enter symbol give just a number "7" on the display, but don't enter the frequency.

By the way, another thing I dislike is that there is just one darkmode screen with not very nice (mostly just white) letters/numbers.

73
Udo, HB9ERD

HB9ERD, 2025-03-14:
I re-flashed to the chinese Firmware (TEF6686_ESP32_2.0.15mod16-20250218), but here, the 7ers come also with every button pressed.

PE5PVB, 2025-03-14:
Switch off and on recovers the buttons

HB9ERD, 2025-03-14:
Ok, thanks, this works (also on the chinese firmware, where the buttons firstly also did not work.
I will stay on the chinese fw for the time being, as I like more the colors there.

closed 2025-03-16 by PE5PVB