PE5PVB/TEF6686_ESP32 #814: Autoscan doesn't log to RabbitEars when memory channels only.

Chuterix, 2025-03-15:
The TEF autoscan doesn't upload to RabbitEars FM bandscan map (https://rabbitears.info/tvdx/fm_all_tuners), where my tuner is at https://rabbitears.info/tvdx/fm_one_tuner_map/208

I've done the autoscan on March 14, with 2 memory channels to demonstrate tuned to extremely strong locals with perfect RDS. There is no slight delay when cycling back to the start memory channel, which means the autolog is not occurring.

kb7wox, 2025-03-15:
This issue doesn't make sense if the FM DX options menu is what the user should be using for the "RabbitEars" band-scan.

PE5PVB, 2025-03-16:
Maybe @kb8u-gqrx can have a look at this?

kb8u, 2025-04-24:
pull request for this bugfix has been sent.