PE5PVB/TEF6686_ESP32 #819: After using touch, it is easy to blur the screen when tuning.

LYMarvin, 2025-03-21:
Compared with the previous version, the v2.20 version is easy to use the portable touch screen when tuning.

PE5PVB, 2025-03-21:
What do you mean? I don't understand your issue.

LYMarvin, 2025-03-22:
I'm sorry, it was my fault for not expressing myself clearly. 
When using version v2.20 for tuning, the display screen frequently experiences flickering/distortion! I had to revert to v2.11.27. With v2.11.27 version for tuning, there is basically no screen flickering.

PE5PVB, 2025-03-22:
There is no difference on this between the versions

LYMarvin, 2025-03-22:
The difference in my test results is quite obvious.

LYMarvin, 2025-04-03:
here:  TEF66868.ESP32 v2.20 Test Video
https://youtu.be/ZSSsF3ET04c

LYMarvin, 2025-04-03:
TEF66868.ESP32 v2.11.27 Test Video
https://youtu.be/v1MOAyNdE8w

PE5PVB, 2025-04-03:
I really have no idea. And you are the only one having this problem. Maybe recompile the firmware and upload it to your radio. As from the look it seems to be a DIY kit. Maybe try another ESP32 board.

closed 2025-06-06 by PE5PVB