PE5PVB/TEF6686_ESP32 #821: Error the COM-port on update

vorobjovhome, 2025-03-28:
According to the radio manufacturer's instructions, I installed the driver for CH340. I downloaded firmware release 2.20. I do everything according to your instructions, but formatting does not start, a port error occurs and a message appears:
" Error uploading, Please check the COM-port and radio for download state. Press any key to exit the update tool.
I tried everything, installed the old version of the CH340 driver, changed the USB cable, and tried to update from another computer.
My radio does not have a BOOT button, but I tried to specify during the update that the button is there and that the button is not. All is useless. Even formatting does not start.
Maybe, to begin with, you need to reset all the settings in the radio to factory settings?

[image]

[image]

[image]

[image]

[image]

andimik, 2025-03-28:
Have you tried different USB cables?

Another PC?

Do you have a Linux Boot DVD? But this requires some background knowledge.

vorobjovhome, 2025-03-28:
1. Yes, I have tried different USB cables, including high-quality Ugreen ones.
2. Yes, I tried installing from different computers. One computer with Windows 11 and the newest version of the CH340 driver, and the other with Windows 7 and the old version of the CH340 driver. The result is the same.
3. I have a computer with Ubuntu. If someone told me, I would try.
      Now I am more inclined to think that I need to reset the radio settings to the default state. But since my radio does not have a reset button, I do not know how to do this?

lvxc88, 2025-03-28:
Do not enter n in this step(step 2), try entering y instead

vorobjovhome, 2025-03-28:
In step 2 I have already tried different options, and pressed "Y", and pressed "N", and pressed ENTER without any letters. In the end, the COM port error appears in any of the options.

lvxc88, 2025-03-28:
Did you ensure that the radio entered download mode before pressing Y? The prompt is that it has not entered the download state

vorobjovhome, 2025-03-29:
My radio does not have a RESET button. When updating the firmware, the display itself goes out several times. Apparently, the radio is designed to automatically go into boot mode. Or am I missing something? How do I put the radio into boot mode if it does not have a reset button?

giovenri, 2025-04-01:
I received this radio today, the desktop version with the small remote control included (FW 2.11 TEF6686 Lithio Patch v.102). I have the same problem, I updated the CH340 drivers, I changed the cables, I tried to press the 9 button on the remote control which sends the radio into boot mode (the radio turns on with the screen off), I followed the instructions point by point but at the end I get the warning that it came out at the OP. Otherwise the radio works fine but it's a pity I can't upgrade it to 2.2.

vorobjovhome, 2025-04-02:
So, the problem is not random, but has a systemic nature. The main thing is to understand the reason. I think that you need to reset the radio firmware to factory settings and then do the update. If this does not help, then I don’t know. Probably, there is some other way to flash, for example, via Linux. I would be grateful if someone gave instructions on how to do this.

giovenri, 2025-04-02:
Is there a way to do a factory reset on this radio, maybe a key combination? I don't see any entries in the menu for a reset nor any physical buttons. Thank you for your help.

andimik, 2025-04-02:
When (!) do you switch it on? You must not do it before the script says so.

giovenri, 2025-04-02:
I tried every which way, switch on when the script say so or before. Nothing changes.

PE5PVB, 2025-06-06:
This is not a firmware issue. Please use Discord to find a solution.

closed 2025-06-06 by PE5PVB