PE5PVB/TEF6686_ESP32 #823: Auto search in AM is not stopping on strong signals.

radiola2, 2025-03-29:
Auto search in AM is almost useless as it does not stop on strong signals. This is on a DP-666.

PE5PVB, 2025-03-31:
Did you change the search sensitivity in the menu?

radiola2, 2025-04-03:
I have tried different settings of the search sensitivity but it does nothing to cure the problem. 

I did not pay much attention to the tuning indicator before, but now I have come to the conclusion that my problems are related to frequency offset.

My DP-666 shows the red arrow that points to the right on almost all shortwave stations. That means my radio is a little off in frequency. The Auto tune function only finds stations that has a bit of frequency offset on the lower side on SW.

For example it stops on Radio 208 that is on 5969,977 kHz according to Twente WebSDR. That station also lights up the green "tuned" dot. It doesn't stop on WMR on 5929,979 khz or stations that are spot on the frequency like Romania that is very close to 9700,000 kHz even if it is very strong.

It seems like the frequency tolerance of the auto search function should allow much more frequency offset than it does now. Also it would be good to have a setting for adjusting frequency offset.

I will do some tests with frequency offsets with a signal generator tonight at work.

radiola2, 2025-04-04:
I set my radio to 10000 kHz and observed the behaviour at different signal offsets.

Signal generator at less than 9999.59 kHz: right arrow (pointing left), auto tune skips.
9999.59 - 9999.89 kHz: green "tuned" indicator lights up, auto tune stops like it should.
9999.89 - 9999.98 kHz: no tuning indicator, auto tune skips. Is this indicator gap a bug in the ShowOffset() function?*
More than 9999.98 kHz: left arrow (pointing right), auto tune skips.

The "tuned" window is only +-0.15 khz which is too narrow. My opinion is that it should be at least +-0.5 khz.

*Maybe the ">" in row 3502 in TEF6686_ESP32.ino should be ">=" ?

PE5PVB, 2025-04-07:
>*Maybe the ">" in row 3537 in TEF6686_ESP32.ino should be ">=" ?

Can you test it? Looks like there is some drift on your crystal.

radiola2, 2025-04-08:
I have tested it and it seems to have fixed the tuning offset indicator. The gap is gone.

Is it possible to compensate for the crystal offset in software, or do I have to do some hardware work?

radiola2, 2025-05-14:
Now I have connected a 39 pF capacitor in series with the 9.216 MHz crystal. The frequency on shortwave is now spot on and auto search is working great!

Still it would be good if this was possible to adjust in software.