PE5PVB/TEF6686_ESP32 #839: AF is not always displayed when there is only one frequency in the list

LucasGallone, 2025-04-10:
Hi,

As the title says, I have observed that the AF list is not always displayed on signals with a list containing only one frequency. Sometimes it does, but most often it doesn't. The TEF shows "No AF available" instead.

The list would be eventually displayed in case of a false positive (A frequency that doesn't exist in the list, being displayed by the receiver, in addition to the real one).

This is happening on the station I offer technical assistance for. We are using Thimeo StereoTool for the RDS encoding, the AF list is totally normal with only one frequency, we don't use anything unusual on the configuration side for that. The issue is also observed on locals that do not use StereoTool for the decoding, so I assume the RDS encoder type has no impact on this issue.

Sometimes that is really worth to note: AF lists with a single frequency are correctly displayed on the webserver. This is only happening on the TEF display (I use the latest version).

Thanks in advance for your investigation.

PE5PVB, 2025-04-10:
What is the frequency in that AF list?

PE5PVB, 2025-04-10:
So the AF frequency is the frequency of the station itself? If so, try to disable AF sorting.

LucasGallone, 2025-04-10:
Exactly. I already have the AF auto sorting feature disabled. To enable it doesn't change anything.

LucasGallone, 2025-04-11:
This aside, I wonder how the webserver is able to display the AF list when there is only one frequency, while the TEF6686 display is not able to do so...? As far as I can remember, I was able to see the lists with a single frequency on the previous versions without any issue.

lxsxl, 2025-09-03:
I checked it again: If there is only one frequency transmitted, it will be shown if AF are unsorted. In the other case (AF sorted), it will not be displayed. But IMHO we should do it the same way and show it there as well.