PE5PVB/TEF6686_ESP32 #849: Compilation error v2.20

Haranochi, 2025-05-04:
Hi, 

I downloaded the source files for the latest v2.20 update. 

I noticed a new variable that doesn't exist in the previous version. It is: #include <SPIFFS.h>
 
 When I run the sketch verification in the Arduino IDE, it displays the following error:
 
**Sketch uses 3149681 bytes (100%) of program storage space. Maximum is 3145728 bytes.text section exceeds available space in board.**

I understand the error is that the partition scheme is not correct. 

The software installed on my computer is: Arduino IDE 1.8.19. 
Esp32 board manager is v2.0.17. 
Partition scheme to Huge APP. (3MB APP / 1MB SPIFFS). 

My TEF6686 receiver is built by me following the rev. B schematic from the PE5PVB website.
 I currently have version 2.0 software installed, and it works very well. 
Has anyone had this problem? Is there anyone who can help me solve it? 

Thank
José María

closed 2025-05-08 by Haranochi

andimik, 2025-05-09:
Maybe it would be good if you could write how to solve it.

Haranochi, 2025-05-10:
Sorry for the delay in responding.
The solution to the question I posed is not mine.
The problem was solved with the modification PE5PVB made to the **constants.h** file, on May 7th (Removed dead file link).
Recompiling the files again did not reveal any errors that would stop the compilation.
The software developer was the one who discovered the problem.

José María