PE5PVB/TEF6686_ESP32 #851: FM-DX Logbook failure: "Failed to open logbook"

LucasGallone, 2025-05-09:
Hi,

Since one of the recent versions, the logbook feature seems to work irregularly.
Most often, it would lead to absolutely no problem and the CSV file is complete. But sometimes, when interrupting the logging process and running a new logging session (without clearing the memory from the first session), then accessing the web interface to collect the CSV file, I would get this error message: "Failed to open logbook".

Rebooting the receiver allows me to access the web interface again, but the CSV has missing entries from the second session. It happened yesterday again, my TEF was running as "Logger" for nothing on about 80km...

I doubt this is a problem from my receiver itself since it started to happen on a recent version. I can't remember which one unfortunately.
Please let me know if you need more details about the issue.

[image]

dxsphere, 2025-05-23:
I'm getting this error as well.

LucasGallone, 2025-06-06:
Thanks for the fix!

PE5PVB, 2025-09-01:
I had to revert this fix because it caused several problems. There is no easy way to fix this. It seems that the filesystem can get corrupted in specific situations. There is no easy fix for this, because the file system is also used for the USA callsign database. I can't switch to a better file system because of lack of memory.

So, no fix possible.

closed 2025-09-01 by PE5PVB