PE5PVB/TEF6686_ESP32 #862: Squelch box cropping digit from number

tomasz-tef, 2025-07-20:
When you choose dBm and try to squelch it doesn't fit values in the box, cropping to only two of them (e.g. -111 becomes -11); I guess the same problem is on different themes/fonts. Mine happened on theme Essence and Aura2 font.

MCelliotG, 2025-07-20:
Please show an example so I can provide a fix.

MCelliotG, 2025-07-21:
Ok pushed the commit.[image]

PHil784, 2025-08-22:
Hello,
I found an issue with the SQ. Below -99 dBm (-100), the screen shows -10.
There is only 2 digit instead of 3.
Theme Goldbrite
POlice Aura2
Software version TEF 2.20

PE5PVB, 2025-09-01:
This one is fixed and will be included in a next beta.