PE5PVB/TEF6686_ESP32 #871: Size too big for memory?

lxsxl, 2025-09-01:
I tried to update to the new version, updated the board-manager on Arduino IDE to 3.3.0, compiling works but upload to the device is not possible:

"Der Sketch verwendet 3150579 Bytes (100%) des Programmspeicherplatzes. Das Maximum sind 3145728 Bytes.
Globale Variablen verwenden 64008 Bytes (19%) des dynamischen Speichers, 263672 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
Der Sketch ist zu groß; unter https://support.arduino.cc/hc/en-us/articles/360013825179 finden sich Hinweise, um die Größe zu verringern.
Textbereich überschreitet den verfügbaren Platz auf der Platine

Compilation error: Textbereich überschreitet den verfügbaren Platz auf der Platine"

The settings are as follows:

[image]

Do you have a good advice to get it successful? Thanks!

PE5PVB, 2025-09-01:
Did you update the tft_espi library and set the confitguration right?

lxsxl, 2025-09-02:
Thank you for your advice, update is done! 👍

lxsxl, 2026-02-16:
Actually I experience the same issue as above. Settings are correct, so I hope you have a good idea for me :-)

Der Sketch verwendet 3429186 Bytes (109%) des Programmspeicherplatzes. Das Maximum sind 3145728 Bytes.
Globale Variablen verwenden 66012 Bytes (20%) des dynamischen Speichers, 261668 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
Der Sketch ist zu groß; unter https://support.arduino.cc/hc/en-us/articles/360013825179 finden sich Hinweise, um die Größe zu verringern.
Textbereich überschreitet den verfügbaren Platz auf der Platine

Compilation error: Textbereich überschreitet den verfügbaren Platz auf der Platine